upsidedownlabs/Chords-Arduino-Firmware #1: Frimware code issue

bushra-br, 2024-11-26:
Hello, I faced a problem with the arduino frimware code. I uploaded the code to my arduino but the website didn't work with me. I connect my Arduino UNO R4 to the laptop, then I chose Arduino UNO R4 COM when I was connecting but the name of the device doesn't appear and I am not able to visiualize any signal at the website.

Also, when I send WHORU in the serial it gives me UNO-R4, but when I send START it stops receiveing.

[image]
[image]

bushra-br, 2024-11-28:
Hello, I didn't find Arduino UNO R4 WIFI which I am using in the source code of the website. 
[image]

lorforlinux, 2024-12-06:
@bushra-br Thank you for letting us know about the issue, we have fixed the sketch for Arduino UNO R4 WiFi. Chords Web will add Arduino UNO R4 WiFi under BoardList very soon, until then you can use Chords-Python to test everything.

New firmware: https://github.com/upsidedownlabs/Chords-Arduino-Firmware/blob/main/UNO-R4/UNO-R4.ino
Chords-Python: https://github.com/upsidedownlabs/Chords-Python

Amanmahe, 2024-12-06:
@lorforlinux @bushra-br i have created a PR that adds support for Arduino UNO R4 WiFi under Chords-Web.

https://github.com/upsidedownlabs/Chords-Web/pull/58

lorforlinux, 2024-12-06:
@Amanmahe Thank you for your contribution, I have merged it.
@bushra-br Please flash new firmware and try it again with Chords-Web it should work now. 

New Firmware: https://github.com/upsidedownlabs/Chords-Arduino-Firmware/blob/main/UNO-R4/UNO-R4.ino
Chords-Web: https://chords.upsidedownlabs.tech/

lorforlinux, 2024-12-28:
@bushra-br I would like to close this issue If you can confirm we have resolved your issue.

bushra-br, 2025-01-03:
The issue has been solved. Thank you so much.

closed 2025-01-14 by lorforlinux