upsidedownlabs/Chords-Arduino-Firmware #2: Firmware Code issue

Z9CK, 2025-01-04:
I have an Arduino Uno R3 board, and I have uploaded the firmware for the Arduino Uno R3 from the github. When I try to visualize on the website, I can't see the waveform. But the device is connected. A pop-up is showing up
[image]
and when i try to send command using serial monitor, i get the following error as "UNKNOWN COMMAND."
[image]

lorforlinux, 2025-01-04:
Hey, please try UNO-CLONE code.

Z9CK, 2025-01-04:
yes, i have uplaoded the code of arduino clone and when i send the command as "WHORU", I get the value as clone, but when i send the command "START" i get random unreadle values. and even in the chords web, i dont get the graph

lorforlinux, 2025-01-04:
Please share photo of your board, I think we know what's the issue. We also
have a UNO clone board which only works at 115200 but it's VID PID matches
real arduino thus it causes issues with Chords.

Faariq-Siddiqui, 2025-01-04:
> Please share photo of your board, I think we know what's the issue. We also have a UNO clone board which only works at 115200 but it's VID PID matches real arduino thus it causes issues with Chords.
> […](#)

[image]
[image]
[image]
[image]

This is the Arduino we are using. Please help

lorforlinux, 2025-01-14:
@Z9CK @Faariq-Siddiqui The board you are using has a USB-UART issue and can only be used with Chords-Python.

1. Upload `UNO-CLONE` Arduino firmware to your Arduino clone board.
2. Use chords python with the command `python chords.py -b 115200 --lsl`
3. Use brain vision LSL to view your LSL stream data.

Note: Install all the dependencies for chords in Python before you start using the scripts.

manobendro, 2025-01-18:
> I have an Arduino Uno R3 board, and I have uploaded the firmware for the Arduino Uno R3 from the github. When I try to visualize on the website, I can't see the waveform. But the device is [...]

- Maybe the baud rate for the USB-to-UART chip on your Arduino board is very high.
- The status message shows a baud rate of 230400.
[image]

- The second issue might be caused by attempting to access the same COM port simultaneously through different sources, such as chords or the Arduino Serial Monitor. (A COM port can only be accessed by one source at a time.)

lorforlinux, 2025-03-20:
All clone boards should work fine with new firmware and chords application thus I am closing this issue now.

closed 2025-03-20 by lorforlinux