upsidedownlabs/Chords-Arduino-Firmware #3: About the data recorded

bushra-br, 2025-02-15:
Hello, I have a question about the data recorded from chords. Is there a way to get the data as channel per time not per counter. Or is there a basis for how the counters divided based on time? Because each time I record a data the number of counters (which I mean number of rows in the excel file) is different even if the recording time is same. 

[image]

lorforlinux, 2025-02-28:
Hey @bushra-br, if you are using a device running at 250sps the time between two points is 4ms whereas for a 500sps device, it will be 2ms. The time between samples in ms = 1000/sampling rate. If the difference is only ±40 samples for 10 - 30min data then it's due to the clock drift of the device. Your Arduino creates its clock but it's not very accurate in providing.

lorforlinux, 2025-03-20:
@bushra-br Did we manage to resolve your issue?

lorforlinux, 2025-04-18:
Due to no response to my last message, I have to close this issue.

closed 2025-04-18 by lorforlinux